sparkfunX/Artemis_Global_Tracker #3: GPS and modem at the same time

eried, 2020-10-01:
I know is not possible, but is there is an easy hack to make this possible? (cutting some traces, adding an extra power regulator to that new rail? etc?)

Has anyone tried? This limitation is quite dumb i.e. if you want to constantly have ring alerts while checking the position with some logic on the artemis module.

PaulZC, 2020-10-02:
Hi Erwin (@eried ),
The only way to do this with the AGT would be to add a second GPS module, connected via Qwiic.
One of the design criteria for the AGT was to keep the mass as low as possible. Antenna sharing saves close to 15g.
Best wishes,
Paul

closed 2020-10-02 by PaulZC

eried, 2020-10-02:
Thanks for the answer Paul, just to confirm as a rule of thumb I just should always call (in the examples) `gnssOFF()` before `digitalWrite(iridiumPwrEN, HIGH);` or `digitalWrite(iridiumPwrEN, LOW);` before `gnssON()` to be 100% on the safe side. Correct?

PaulZC, 2020-10-03:
Hi Erwin (@eried ),
To disable the Iridium modem, it is best to disable both the modem itself _and_ the supercapacitor charger:
https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Software/examples/Example1_Blink/Example1_Blink.ino#L53-L56
To disable the GNSS, yes, ```GNSSoff()``` will suffice:
https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Software/examples/Example1_Blink/Example1_Blink.ino#L57
Good luck with your project!
Best wishes,
Paul