sparkfunX/Artemis_Global_Tracker #4: Avoid reset when re-connecting via serial?

eried, 2020-10-07:
Hi, is there is any way to prevent the device from resetting if I close and open the serial port?

I see that the reset is:  ` resetResult.bEXTStat // Serial.println("External reset");`

PaulZC, 2020-10-07:
Hi Erwin,
We use the serial reset to put the Artemis into bootloader mode. It is possible to disable this but you would no longer be able to upload new code.
Best wishes,
Paul

closed 2020-10-07 by PaulZC

eried, 2020-10-08:
Thanks for the answer, but there is no easy way to disable the reset in a temporal way and just require pressing the reset button when wanting to upload new code?

Since this is part of a global tracker, it would leave using the eeprom as the only option (i.e. to prevent losing messages received that reside on ram waiting to be downloaded via serial later)

PaulZC, 2020-10-08:
Hi Erwin,

I'm not sure what your end application is, but, yes, fair point, I can see why you would want to avoid the reset.

Message storage in EEPROM is certainly possible. The AGT settings are all stored in EEPROM in Example16.

Another way would be to connect a micro SD card via the SPI breakout pins. One of [these](https://www.sparkfun.com/products/544) should work nicely.

I really do not recommend it, and have never tried it, but you should be able to disable the reset by removing some of these components:
[image]

Regards,

Paul