sparkfunX/Artemis_Global_Tracker #5: power consumption

jerabaul29, 2020-11-06:
Hi,

I see that you have a "low power" sketch and several mentions of "low power" uses, including recommending some batteries in the hardware overview.

Do you have some actual measurements of how much energy is typically used for transmitting / receiving a message, and the typical current draw of the uC in deep sleep as for example in the https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Software/examples/Example12_TestLowPower sketch?

And / or said otherwise: if using 3 AA lithium batteries, and transmitting the GPS position once every 3 hours and doing nothing else (ie deep sleep for 3 hours, getting a GPS reading, sending it through Iridium), how long can I typically expect the module to work before the battery is empty?

PaulZC, 2020-11-07:
Hi JR (@jerabaul29),
Thanks for your interest in the Global Tracker.
I've just done a quick test and the sleep current is approximately 250uA. That's probably higher than usual as I haven't used this board for a while and the RTC battery will be charging. I'll leave it for a few hours and will check it again.
The second part of your question is more complex. It of course depends what batteries you are using and how often you are transmitting. The bulk of the power goes into charging the super capacitors for the Iridium transmit. I did use an earlier - less efficient - version of the tracker to track a balloon flight:
- Using three AA Energizer Ultimate Lithium
- During the flight the beacon was transmitting every 3 minutes and sent 140 messages
- The payload ended up in a tree and it was weeks before we could pick it up
- Transmitting twice per day for a month, it sent another 50 messages
- We then set it to transmit once per day and it sent another 17 messages before we finally collected it
- Battery voltage at the end was just under 4.0V. It was still quite happy.
- The flight ended up in a tree in Austria in March (2019). Daytime temperatures were around 0C. Heavy snow was the main thing from preventing us from collecting it.

The core design is very robust. We've had a lot of fun with it over the years.
There's a paper here that you might enjoy:
http://arxiv.org/abs/1904.04321
Best wishes,
Paul

jerabaul29, 2020-11-07:
That sounds excellent, many thanks for the technical details and the ArXiv link :) .

PaulZC, 2020-11-07:
Looks like the RTC battery is indeed charging. Current draw has dropped to around 220uA after ~1 hour.

PaulZC, 2020-11-07:
This is with the power LED disabled - of course! Open the split pad on the rear of the PCB to disable it.

PaulZC, 2020-11-09:
Hi JR (@jerabaul29) - I have checked the figures and have added more information here:
https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/GlobalTracker_FAQs

closed 2020-11-09 by PaulZC

jerabaul29, 2020-11-11:
Hi again!

I was thinking a bit about the power consumption while sleeping. It is quite a bit higher than the pure Apollo3 MCU. Do you see some ways to further reduce the deep sleep power consumption?

PaulZC, 2020-11-11:
Hi JR,
There are lots of very small current draws that add up to give the 220uA. The ~55uA quiescent current draw of the AP2112K regulator being just one. The ~30uA current draw of the Artemis being another. Beating the 220uA down below (say) 100uA would be a time-consuming project....
Best wishes,
Paul