sparkfunX/Artemis_Global_Tracker #6: power in: minimum voltage

jerabaul29, 2020-11-06:
In the documentation:

https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/Hardware_Overview

you mention that one can power the module with:

"an external solar panel or battery pack (6V maximum)".

There is not indication of the minimum voltage needed. I suppose it is min_voltage_in = 3.3V + loss_regulator ? What is the typical loss in the regulator that needs to be factored in in choosing the battery?

PaulZC, 2020-11-07:
Hi again JR (@jerabaul29),
It depends... At 200mA current draw, the AP2112K regulator has a drop-out of approx. 75mV. So, 3.375V is the official answer.
However, you can push it lower than that. Yes, the regulator will start to drop out, but the Artemis will be quite happy down to 1.8V. The super capacitor charger needs a minimum of 2.8V. The ZOE-M8Q is OK down to 2.7V.
The current draw does increase as the regulator goes into drop-out so you need to be aware of that.
Energizer Ultimate Lithium batteries will deliver around 1.1V per cell when very cold (around -40C).
The design includes a low voltage reset supervisor which helps - mostly for solar power where the voltage can rise very slowly at sunrise.
Best wishes,
Paul

jerabaul29, 2020-11-07:
Ok, many thanks for the technical details and the confirmation :) . I think for the Arctic and Antarctic (which I am interested in :) ). I would then use 4 LiTi batteries just to be on the safe side :) .

PaulZC, 2020-11-07:
Just make sure that they can deliver enough current. Some Lithium chemistries can only deliver a few mA continuous draw.
Saft LSH batteries are good - but don't go higher than 6.0V or bad things will happen...
A fresh Energizer Lithium AA or AAA will deliver around 1.8V. It soon droops, but you do need to be aware.

PaulZC, 2020-11-09:
Hi JR (@jerabaul29) - I have checked the figures and have added more information here:
https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/GlobalTracker_FAQs

closed 2020-11-09 by PaulZC