sparkfunX/Artemis_Global_Tracker #8: antenna

jerabaul29, 2020-11-08:
It sounds from the documentation:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Documentation/Hardware_Overview/README.md

like the "recommended" antenna is the following:

https://www.sparkfun.com/products/15789

- can you confirm if it is well this Sparkfun antenna that should be used?

- any active antenna that also could be used? In the context I look at it is a good idea to use an active antenna.

PaulZC, 2020-11-08:
Yes, that antenna will work fine. It is optimized for Iridium but provides good signal in the GNSS bands too. There are antennas which are tuned for both GNSS and Iridium but they are more expensive.
You can't use an active antenna. The Iridium transmit would destroy the active pre-amp. GNSS receivers are so sensitive these days that you really don't need an active antenna unless you are in a poor signal area.

closed 2020-11-08 by PaulZC

jerabaul29, 2020-11-08:
Thanks! Same, wonder if this should be made sticky and or added to the doc :)

PaulZC, 2020-11-09:
Thanks JR (@jerabaul29) - good suggestion. Added here:
https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/GlobalTracker_FAQs