sparkfunX/Artemis_Global_Tracker #10: bluetooth

jerabaul29, 2020-11-09:
If I understand well the Artemis uC has bluetooth abilities, right? :) So do you think a method of this kind:

https://learn.sparkfun.com/tutorials/artemis-development-with-the-arduino-ide/bluetooth-led

Could be used to start / parametrize the tracker? That would be especially attracting for applications where a fully sealed IP 68 or higher box is built, and you need to start it at the deployment time without the need to open it / fiddle with switches etc :) .

PaulZC, 2020-11-09:
Hi JR,
The AGT will support BLE - but not at the moment. Upgrading to v2 of the Apollo3 core is on the things-to-do list!
Best wishes,
Paul

jerabaul29, 2020-12-06:
Just being curious - this would mean shifting from an Arduino "main loop" programming paradigm to a "threaded RTOS" one, right? That would be interesting to follow :) .

razvandragomirescu, 2021-07-29:
Is Bluetooth going to "just work" after the upgrade to v2 of the Apollo3 core? I have not updated yet because the AGT Iridium examples stopped working when I tried that. Is there a particular version that supports both BLE and Iridium?

PaulZC, 2021-07-30:
Hi @razvandragomirescu ,
The upgrade to v2 of the Apollo3 core is still on the to-do list. We recently upgraded OpenLog Artemis to use v2.1.0 of the core and are now working on adding ways of accessing the sensor data over BLE. The AGT is next.
So, to answer your question, yes, the AGT will support both BLE and Iridium when the upgrade is complete. You will be able to use the standard ArduinoBLE library to advertise services and add characteristics etc..
Thank you for your patience,
Paul

PaulZC, 2021-07-30:
Hi @jerabaul29 ,

I would like to start a discussion about how the BLE functionality on the AGT should behave / perform. Please add your thoughts to Issue #34.

Many thanks,
Paul

jerabaul29, 2021-08-02:
Sounds good. Closing this in favor of 34 then :) .

PaulZC, 2021-08-02:
Thank you JR,
Best wishes,
Paul

closed 2021-08-02 by PaulZC