sparkfunX/Artemis_Global_Tracker #11: supercapacitor

jerabaul29, 2020-11-30:
Hi,

I just bought a couple of the module trackers from Sparkfun. However, it seems that they come without the supercapacitors. This is a bit sad, as these are not mentioned in the list of "essential components" (I will write to sparkfun to get this added to the list), I guess I will have to place an additional order...

One thing about these capacitors: it is written that these should be 10F, 2.7V. I cannot find supercapacitors with such speccing from Sparkfun, the closest I get is these, but I guess they will not like at all the 2.7V part of it :( 

https://www.sparkfun.com/products/746

My question is then: which super capacitor did you use / where did you source it? :)

Cheers,

JR

PaulZC, 2020-11-30:
Hi JR (@jerabaul29 ),
The AGT comes equipped with 1F supercapacitors. You only need the extra 10F supercapacitors if you change the charge current to 60mA.
Please see the documentation here:
https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/Hardware_Overview#iridium-9603n
Best wishes,
Paul

PaulZC, 2020-11-30:
Hi JR (@jerabaul29 ),
I've added this to the FAQs too:
https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/GlobalTracker_FAQs#Do-I-need-the-extra-10F-supercapacitors
Best wishes,
Paul

closed 2020-11-30 by PaulZC

jerabaul29, 2020-11-30:
Ah, I had missed that, many thanks!