sparkfunX/Artemis_Global_Tracker #12: using own regulator

jerabaul29, 2021-01-06:
I would like to bypass the low drop power regulators that are present on the board and use some step up-step down converters instead due to the technology of batteries I want to use.

Would it be ok to use this regulator:

https://www.pololu.com/product/2561

"This 3.3 V boost (step-up) voltage regulator generates higher output voltages from input voltages as low as 0.5 V, and it also automatically switches to a linear down-regulation mode when the input voltage exceeds the output. This makes it great for powering 3.3 V electronics projects from 1 to 3 NiMH, NiCd, or alkaline cells or from a single lithium-ion cell."

And use it to directly feed to the 3.3V pin?

jerabaul29, 2021-01-06:
(i.e.:

- is it enough to only provide a well-regulated 3.3V input to the whole board?
- if providing a well-regulated 3.3V input to the board, where to feed it so that we bypass the regulators and avoid their power drops?

jerabaul29, 2021-01-06:
From reading the schematics at https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Documentation/Hardware_Overview/README.md it looks like:

- USB, Solar and Lipo are all fed into the VBUS through diodes
- VBUS is fed (through a cuttable pad to measure power usage) to VIN
- VIN is used as input to the low dropout power generator, to generate the 3.3V supply used by everything on the board (except the USB interface of course).

That would speak in favor of simply feeding the output of the buck converter into the GND and 3.3V pins, for example the one on the side of the SDA / SCL pins. Can you confirm that? :)

jerabaul29, 2021-01-06:
One more thing: if this understanding is right, can you update a bit the "Power Options" section:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Documentation/Hardware_Overview/README.md

saying that there is more info in the FAQ at https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Documentation/GlobalTracker_FAQs/README.md#What-voltage-does-the-AGT-need

And in the FAQ, can you make it clear that there are in addition to the recommended modes of powering 2 "unsafe / expert modes" ways to power: either 1) through the VBUS in which case the polarity protection is dropped but not the low dropout regulator (i.e. input should be 3.4V to 6.0V), or 2) directly through the 3.3V common power rail, in which case there will be no polarity nor voltage protection, and the only acceptable voltage is EXACTLY 3.3V, as provided for example by a step up / step down converter?

PaulZC, 2021-01-07:
Hi JR (@jerabaul29 ),
I hope your project is going well.
I've added the link to the FAQs from the Hardware Overview as requested.
Sorry, no, I won't be adding any notes about unsafe / expert modes for powering the board. You are welcome to try the buck/boost regulator and it will probably work fine, but it is at your own risk.
Best wishes,
Paul

closed 2021-01-07 by PaulZC

jerabaul29, 2021-01-07:
Ok, no worries. Thanks for confirming it should work fine.

PaulZC, 2021-01-07:
"probably"
;-D

jerabaul29, 2021-01-07:
(btw, to give you some context: thanks again for your product, you came up with a great design. We had a very small scale test here at the Norwegian Meteorological Institute a few weeks ago and were happy with the results, so we just ordered a new batch of 10 of these for a small pilot study. We will deploy these as sea ice drifting trackers in the Arctic. We expect temperatures down to -40, and some preliminary tests showed that even Li batteries had a bit of trouble then, which is why we need to add a step up / step down regulator).

jerabaul29, 2021-02-11:
(fyi: this works fine in the cold, we are getting messages from our first instruments deployed 78N :) ). I mistakenly put some of the step-up / step-down on VBUS instead of 3.3V, but seems to work fine too.