sparkfunX/Artemis_Global_Tracker #16: using the core 2.0.5: anything to think about regarding pin numbering

jerabaul29, 2021-02-11:
I am re-writing a bit of firmware so that it fits my needs better. I am using the MbedOS core, v 2.0.5, from there: https://github.com/sparkfun/Arduino_Apollo3 .

Anything to be careful of / that may be problematic porting, and that I should think about? Or should everything just go fine? I am in particular worried if some "pin number matching" between the physical pins of the MCU, and the pins numbers in the new core, may be different, and / or which "board variant" to use to make sure all pin numbers are correctly mapped.

PaulZC, 2021-02-11:
Hi JR,

The pin numbers _should_ be the same - but you're entering unchartered territory here! I haven't yet tried to migrate any of my Artemis projects to Mbed.

Best wishes,

Paul

closed 2021-02-11 by PaulZC

jerabaul29, 2021-02-11:
Ok, I will let everybody know here if I face any issues :) .