sparkfunX/Artemis_Global_Tracker #17: confused: is the iridium modem on I2C or serial?

jerabaul29, 2021-02-11:
I am a bit confused about how the iridium modem is connected. I thought from the use of:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/5e40b4283dfdd46b7b90fed9639ad0af667707ea/Software/examples/Example14_SimpleTracker/Example14_SimpleTracker.ino#L20-L23

that it would be I2C, but checking the code it seems like it is actually serial, right?

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/5e40b4283dfdd46b7b90fed9639ad0af667707ea/Software/examples/Example14_SimpleTracker/Example14_SimpleTracker.ino#L72-L74

Checking the schematics it also seems like it is on serial not I2C.

Another question related: any particular reason for doing:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/5e40b4283dfdd46b7b90fed9639ad0af667707ea/Software/examples/Example14_SimpleTracker/Example14_SimpleTracker.ino#L74

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/5e40b4283dfdd46b7b90fed9639ad0af667707ea/Software/examples/Example14_SimpleTracker/Example14_SimpleTracker.ino#L79

Instead of something like:

```cpp
IridiumSBD modem(Serial1, iridiumSleep, iridiumRI);
```

PaulZC, 2021-02-11:
Hi JR,

Correct - it does use Serial. SparkFun_IridiumSBD_I2C_Arduino_Library is the name we gave to our fork of Mikal Hart's IridiumSBD library, to indicate that it also supports I2C - on the [Qwiic Iridium 9603N](https://www.sparkfun.com/products/16394). Mikal's library is limited to serial only. Ours includes extra features and a few bug-fixes too.

The Artemis Serial ports can use different combinations of pins. ```Uart iridiumSerial(1, 25, 24);``` creates a port using the Serial1 hardware with RX and TX on pins 25 and 24. If you use the generic "Serial1" definition, the pins will be different.

Cheers,

Paul

closed 2021-02-11 by PaulZC

jerabaul29, 2021-02-11:
Many thanks, that makes it much clearer :) .