sparkfunX/Artemis_Global_Tracker #18: cannot start the GPS on the core v 2.0.5

jerabaul29, 2021-02-11:
I do not manage to turn on the GPS using the core v 2.0.5.

I noticed that you use ```Wire1``` and not ```Wire``` for talking to the GPS:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/5e40b4283dfdd46b7b90fed9639ad0af667707ea/Software/examples/Example14_SimpleTracker/Example14_SimpleTracker.ino#L342

and in all the following.

However in the core v 2.0.5 I can only use ```Wire```. Is it that there are several I2C ports, and ```Wire``` vs ```Wire1``` are two different ports? In this case, I suppose I would need that the ```Wire1``` gets added to the core v 2.0.5 for it to work, right? Or do you see another workaround?

PaulZC, 2021-02-11:
This example might help?
https://github.com/sparkfun/Arduino_Apollo3/blob/master/libraries/Apollo3/examples/I2C/I2C.ino

jerabaul29, 2021-02-11:
Ah yes you are right, thanks! So I should be able to do something in this kind then, right? :) Will look at it tonight or tomorrow.

https://github.com/sparkfun/Arduino_Apollo3/blob/fbb2eaca27245627bf22b59f43faa232635bcd01/libraries/Apollo3/examples/I2C/I2C.ino#L65-L67

jerabaul29, 2021-02-12:
Mmmh, getting new issues.

From reading the schematics, I am thinking that the pins to use are D9 and D8. So I tried something like:

```cpp
#include <Wire.h> //Needed for I2C to GNSS

[...]

MbedI2C gps_wire(D9, D8);
```

But I get an error:

```
first start GPS I2C port

++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: pinmap not found for peripheral
Location: 0x25053
Error Value: 0xC
Current Thread: main Id: 0x10004BA8 Entry: 0x26975 StackSize: 0x1000 StackMem: 0x10006818 SP: 0x10007754 
For more info, visit: https://mbed.com/s/error?error=0x80010130&tgt=SFE_ARTEMIS
-- MbedOS Error Info --
```

This is quite clear error message, so I also tried:

```cpp
MbedI2C gps_wire(D8, D9);
```

But same kind of error. Any idea how this should be done instead?

jerabaul29, 2021-02-12:
Ok, getting closer: the issue is in how the pins are described. The "D" should be removed, i.e., this works:

```cpp
MbedI2C gps_wire(9, 8);
```

jerabaul29, 2021-02-12:
Ok, back to being stuck... Now my code looks essentially like this (taking away a bit of the usual verbose serial enabling etc):

```cpp
#define gnssEN              26 // GNSS Enable: pull low to enable power for the GNSS (via Q2)

MbedI2C gps_wire(9, 8);
gps_wire.begin();
delay(10);
  
pinMode(gnssEN, OUTPUT); // Configure the pin which enables power for the ZOE-M8Q GNSS
digitalWrite(gnssEN, LOW); // Enable GNSS power (HIGH = disable; LOW = enable)
delay(2000);

    while (true){
        if (gps.begin(gps_wire)){
            Serial.println(F("GPS started"));
            break;
        }
        else{
            Serial.println(F("could not start the GPS"));
            delay(1000);
        }
    }
```

But I get locked in the ```Could not start the GPS``` thing.

- anything I do wrong?
- if not, anybody else who can reproduce the problem?

jerabaul29, 2021-02-12:
Unsuccessful so far after quite a few hours of trying to get this to work. This is my "least complexity example". Anything that looks wrong / are you able to reproduce the problem / any idea how to fix?

```cpp
#include <Wire.h> // Needed for I2C

// #include <SparkFun_Ublox_Arduino_Library.h> //http://librarymanager/All#SparkFun_Ublox_GPS
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>
SFE_UBLOX_GNSS myGPS;

//--------------------------------------------------------------------------------
// from Example 14
// Artemis Tracker pin definitions
#define spiCS1              4  // D4 can be used as an SPI chip select or as a general purpose IO pin
#define geofencePin         10 // Input for the ZOE-M8Q's PIO14 (geofence) pin
#define busVoltagePin       13 // Bus voltage divided by 3 (Analog in)
#define iridiumSleep        17 // Iridium 9603N ON/OFF (sleep) pin: pull high to enable the 9603N
#define iridiumNA           18 // Input for the Iridium 9603N Network Available
#define LED                 19 // White LED
#define iridiumPwrEN        22 // ADM4210 ON: pull high to enable power for the Iridium 9603N
#define gnssEN              26 // GNSS Enable: pull low to enable power for the GNSS (via Q2)
#define superCapChgEN       27 // LTC3225 super capacitor charger: pull high to enable the super capacitor charger
#define superCapPGOOD       28 // Input for the LTC3225 super capacitor charger PGOOD signal
#define busVoltageMonEN     34 // Bus voltage monitor enable: pull high to enable bus voltage monitoring (via Q4 and Q3)
#define spiCS2              35 // D35 can be used as an SPI chip select or as a general purpose IO pin
#define iridiumRI           41 // Input for the Iridium 9603N Ring Indicator
// Make sure you do not have gnssEN and iridiumPwrEN enabled at the same time!
// If you do, bad things might happen to the AS179 RF switch!

void gnssON(void) // Enable power for the GNSS
{
  digitalWrite(gnssEN, LOW); // Disable GNSS power (HIGH = disable; LOW = enable)
  pinMode(gnssEN, OUTPUT); // Configure the pin which enables power for the ZOE-M8Q GNSS
}

void gnssOFF(void) // Disable power for the GNSS
{
  pinMode(gnssEN, INPUT_PULLUP); // Configure the pin which enables power for the ZOE-M8Q GNSS
  digitalWrite(gnssEN, HIGH); // Disable GNSS power (HIGH = disable; LOW = enable)
}
//--------------------------------------------------------------------------------

void setup(){
  //--------------------------------------------------------------------------------
  // from example 14
  // Let's begin by setting up the I/O pins
   
  pinMode(LED, OUTPUT); // Make the LED pin an output

  gnssOFF(); // Disable power for the GNSS
  pinMode(geofencePin, INPUT); // Configure the geofence pin as an input

  pinMode(iridiumPwrEN, OUTPUT); // Configure the Iridium Power Pin (connected to the ADM4210 ON pin)
  digitalWrite(iridiumPwrEN, LOW); // Disable Iridium Power (HIGH = enable; LOW = disable)
  pinMode(superCapChgEN, OUTPUT); // Configure the super capacitor charger enable pin (connected to LTC3225 !SHDN)
  digitalWrite(superCapChgEN, LOW); // Disable the super capacitor charger (HIGH = enable; LOW = disable)
  pinMode(iridiumSleep, OUTPUT); // Iridium 9603N On/Off (Sleep) pin
  digitalWrite(iridiumSleep, LOW); // Put the Iridium 9603N to sleep (HIGH = on; LOW = off/sleep)
  pinMode(iridiumRI, INPUT); // Configure the Iridium Ring Indicator as an input
  pinMode(iridiumNA, INPUT); // Configure the Iridium Network Available as an input
  pinMode(superCapPGOOD, INPUT); // Configure the super capacitor charger PGOOD input

  pinMode(busVoltageMonEN, OUTPUT); // Make the Bus Voltage Monitor Enable an output
  digitalWrite(busVoltageMonEN, LOW); // Set it low to disable the measurement to save power
  analogReadResolution(14); //Set resolution to 14 bit
  //--------------------------------------------------------------------------------

  Serial.begin(115200);
  delay(10);
  Serial.println(F("booted"));

  // on the new core, no Wire1, need to define which pins to use
  MbedI2C gps_wire(9, 8);
  gps_wire.setClock(100000);
    
  Serial.println(F("Powering up the GNSS..."));
  gps_wire.begin();
  gnssON();
  delay(2000); // Give it time to power up

  if (myGPS.begin(gps_wire) == false) //Connect to the Ublox module using Wire port
  {
    Serial.println(F("unable to start GPS"));
  }
  else{
    Serial.println(F("able to start GPS"));
  }

  Serial.println(F("done with setup"));
}

void loop(){
}
```

PaulZC, 2021-08-09:
Hi JR (@jerabaul29 ),

Thank you for your patience!
I am about to release version 2.0 of the AGT firmware / examples - updated and built using v2.1.0 of the Apollo3 core.
If you are going to re-use the examples, please use **v2.1.0** of the core - not v2.1.1. (v2.1.1 contains a new 'feature' which makes I2C communication with the ZOE-M8Q problematic).
As my American colleagues would say, please kick the tyres and let me know what you find.

Very best wishes,
Paul

closed 2021-08-09 by PaulZC