sparkfunX/Artemis_Global_Tracker #20: Adding an I2C sensor - any way to do so in a way that easily turned on / off

jerabaul29, 2021-02-18:
Hi :)

I want to add the BNO080 to this board. As it looks like we cannot at the moment get really low power with the BNO080, I will have to turn it on / off by enabling / cutting power supply to it.

- from the schematics it looks like the Qwiic port is always power on, right (I see no MOSFET controlling power to it)? Or is there a way to turn power on / off on the devices on the Qwiic port?

- another possibility would be to put the BNO on the same power ramp as the GPS, and use a GPS low power mode when it is not in use but the BNO is (the GPS has a low power mode if I remember well? :) ). The challenge it seems is to get access to this power rail. I do not see any obvious place where I could easily solder a couple of wires to get power. Or am I missing something?

- I guess if both previous options are not possible that I will have to just add a separate small PCB with its own MOSFET or similar, right? Or is there another solution I could consider?

jerabaul29, 2021-02-18:
Ahhh, I see a solution: the BNO080 should be happy with everything above 2.4V according to its datasheet Fig. 6.2, right? :) Then I should be able to tap power from the pins of one of the supercapacitors, right? :) Something like:

- putting the iridium modem to sleep
- switching on capacitor charging
- starting to talk to the BNO

Or am I thinking wrongly?

jerabaul29, 2021-02-18:
(of course that would be a hack, but but, good enough for a prototype / proof of concept to me, and we will design a custom board later on if we are allowed to continue in this direction).

jerabaul29, 2021-02-18:
(I would need just to find which is C5 vs C6 I guess to make sure I have the right ground, but I can check that by hand on one of my boards :) If you can tell which one is the one to use, just for double checking, that would be great though :) ).

jerabaul29, 2021-02-18:
Mmmh, no, checked out in more details, that will not work: the datasheet says the voltage on SDA / SCL should not be over Vin. So not possible to power with 2.7V if the logics are at 3.3V I am afraid.

There is one pad under the ATG that I am curious about: the "VIN/3" mini pad in the rectangle around a supercapacitor. I have a hard time figuring out what it is couple to. Is it on one of the MOSFETs?

PaulZC, 2021-02-19:
Hi JR,
Ideally, you need the Qwiic Power Switch but it is out of stock at the moment. [Mouser have 5](https://www.mouser.co.uk/ProductDetail/SparkFun/SPX-16740/?qs=DPoM0jnrROVJN3KjZlloUg%3D%3D).
Correct, the Qwiic power is not switched.
My suggestion would be to hook up the BNO080 to the GPS/MS8607 I2C bus using the breakout pins on the left side of the board. You then need to solder a wire to **3V3SW** to provide switched power. The easiest places to do that would be at Q2:
[image]
Or at the GNSS power LED resistor R16:
[image]
VIN/3 let's the Artemis check the VIN power rail.
[image]
Actually, that's an alternative for you. You could pick up **VIN** power from the 'top' of R24 but you _**must**_ then regulate it to 3.3V before passing it to the BNO080.
[image]
I hope this gets you going.
Paul

jerabaul29, 2021-02-19:
Many thanks for your kind help :) . I will consider either putting it on the GPS LED, or using an additional (I already use 1 to feed power to the board) external pololu regulator, which can be driven on and off through an activate pin.

closed 2021-02-19 by jerabaul29