sparkfunX/Artemis_Global_Tracker #21: charging the GPS backup battery (and keeping it charged)

jerabaul29, 2021-02-19:
@PaulZC I notice that I quite seldom get a "1s hotfix" on the GPS despite the presence of the battery.

I had a look at the schematics. It looks like the only way to charge the battery is to use V_USB, as the diodes would prevent other sources of power to charge it. Is this the right understanding or am I missing something? What is the recommended way to get this small battery fully charged before deployment (and to keep it charged during deployment :) )?

jerabaul29, 2021-02-19:
(would putting a jumper between the 3.3V rail and the + side of the battery help? :) ^^ ).

PaulZC, 2021-02-20:
Hi JR,
Correct, the back-up battery only charges when USB is connected as shown in [this section](https://github.com/sparkfunX/Artemis_Global_Tracker/tree/master/Documentation/Hardware_Overview#zoe-m8q).
If you try different ways of charging the battery, make sure you limit the current appropriately. Connecting directly to 3.3V will release the magic white smoke...
Paul

closed 2021-02-20 by PaulZC