sparkfunX/Artemis_Global_Tracker #22: Examples do not work with any version >v1.2.0

miquelmassot, 2021-04-12:
Since [Arduino_Apollo3](https://github.com/sparkfun/Arduino_Apollo3) got updated to use mbedOS from v2.0.0 the code examples provided do not compile.

PaulZC, 2021-04-12:
Hi Miquel (@miquelmassot ),
Yes, we are aware of this, thank you. This is the same as issues #19 and #18.
We are waiting for issue 2.1.0 of the Apollo3 core to fix these issues.
I will leave this issue open until then as a reminder.
Best wishes,
Paul

a-schneider-fmi, 2021-07-20:
In the mean time, Apollo3 2.1.1 has been released. Compilation works for example 1 but fails for all other examples. When will a compilable tracker firmware be available?

PaulZC, 2021-07-20:
Hi Andreas (@a-schneider-fmi ),

We are working on it! ;-) 

We have just released v2 of the OpenLog Artemis firmware which includes the upgrade to v2.1.0 of the core. Upgrading the AGT firmware is on the things-to-do list - but there are other things above it... I can't promise when the upgrade will happen, but it will be sometime over the next month as we have a project coming up which will need the upgrade.

Best wishes,
Paul

razvandragomirescu, 2021-07-29:
I have it all working with the older core but Bluetooth does not work with that - I understand that after v2 BLE will work but the examples will not and I need to make them both work for the particular project I am working on.

PaulZC, 2021-07-30:
Hi @razvandragomirescu / @a-schneider-fmi / @miquelmassot ,

I would like to start a discussion about the BLE functionality on the AGT should behave / perform. Please add your thoughts to Issue #34.

Many thanks,
Paul

PaulZC, 2021-08-09:
Hi @miquelmassot / @a-schneider-fmi / @razvandragomirescu ,

Thank you for your patience!
I am about to release version 2.0 of the AGT firmware / examples - updated and built using v2.1.0 of the Apollo3 core.
If you are going to re-use the examples, please use **v2.1.0** of the core - not v2.1.1. (v2.1.1 contains a new 'feature' which makes I2C communication with the ZOE-M8Q problematic).
I have not added any BLE examples or functionality - just getting the existing 16 examples updated has taken quite a while - but at least you now have a working skeleton on which to add BLE.
As my American colleagues would say, please kick the tyres and let me know what you find.

Very best wishes,
Paul

closed 2021-08-09 by PaulZC