sparkfunX/Artemis_Global_Tracker #23: problems receiving messages from the satellite

jerabaul29, 2021-04-19:
Hi,

I am using the tracker to send messages through iridium, and this works well.

However, it seems that I have troubles sending messages to the modem.

# What I do to send the message from Rock7

- go to the Rock7 webpage, and to the "Send a message" pane
- write a message and click on send
 
I think this should be correct

# The code I have to receive messages

The code looks something like:

```
    int transmission_status = iridium_sbd.sendReceiveSBDBinary(
      iridium_tx_raw_buffer, tx_buffer_amount,
      iridium_rx_raw_buffer, rx_buffer_amount
    );

    Serial.println();
    Serial.print(F("rx_buffer_amount: ")); Serial.println(rx_buffer_amount);

    if (rx_buffer_amount > 0){
        Serial.print(F("received iridium command: "));

        for (size_t i=0; i<rx_buffer_amount; i++){
            Serial.print(iridium_rx_raw_buffer[i], HEX); Serial.print(F(" "));
        }

        Serial.println();
        rx_message_available = true;
    }
```

# The result I get

I am very puzzled, because it seems that the modem gets to know that there are unsigned chars coming in, and it knows how many, but the transmission status is reported as failure, and the buffer is still filled with 0s when I try to read it. For example, trying to send the message "BOOT", I get:

```
rx_buffer_amount: 4
received iridium command: 0 0 0 0
iridium transmission failed
```

- has anybody else observed problems in this kind?
- any idea if this may be a hardware, software, library problem?

jerabaul29, 2021-04-19:
Activating the debug output it looks like:

```
OK
SBDIX MO code: 0
SBDIX success!
Incoming message!
>> AT+SBDRB

Waiting for response AT+SBDRB

<< AT+SBDRB
[Binary size:4][csum:308]Waiting for response OK

<< 
OK

transmission status [0 is success]: 8
rx_buffer_amount: 4
```

quite puzzled, not sure what the problem is.

jerabaul29, 2021-04-19:
Aaah, sorry, I understand, I had mis-understood the meaning of a parameter to the function. I will check and confirm.

jerabaul29, 2021-04-19:
Ok, so just to confirm. My problem is that I thought that the RX buffer should have size 270 and that the ```rx_buffer_amount``` variable was only for retrieving the number of bytes received.

But inspecting the code, this is not what the ```rx_buffer_amount``` is used to (or not only). It is used:

- first to send in the size of the buffer that is available
- then to retrieve how many byte were written.

Of course, this is a "user error" in how to use the library, nothing to do with the tracker. Still think that the fact that this param is an "inout" is quite tricky and confusing as it is not documented in details either...

But now it is fixed :) Sorry again for the noise.

closed 2021-04-19 by jerabaul29

PaulZC, 2021-04-19:
Hi JR (@jerabaul29 ),

There is some documentation on how to use the ```sendReceiveSBDBinary``` function in the IridiumSBD I2C Library:
https://github.com/sparkfun/SparkFun_IridiumSBD_I2C_Arduino_Library/tree/master/documentation#data-transmission-1

[image]

Best wishes,
Paul