sparkfunX/Artemis_Global_Tracker #27: availability of the tracker

jerabaul29, 2021-05-10:
Hi @nseidle @PaulZC ,

An issue that is maybe a bit "meta". We (I and several colleagues in a number of countries now: Norway, Canada, Australia, Japan) are regularly considering (and actually taking the step of) using the AGT as a basis for some instruments used by our employers in "real full scale" deployments. For example, I have been directly involved in deployment of around 20 instruments based on the AGT so far in the last 6 months or so.

A real issue for us now, when using the AGT in a professional context, is the availability (or lack of availability) of the AGT. For example, I and some colleagues we are considering building a small series (let us say 5 to 10 typically, but if we can convince a manager or two, may be even quite a bit more) of instruments for a deployment from an ice breaker late summer. Seeing that the AGT is out of stock, and not knowing at all when it will arrive and how many units, is really painful to us - that sounds very un-serious when I talk to a manager and says that "sorry, I have no idea if / when / how many electronics boards we will get our hand on, and how many instruments I could deliver, even if you can provide the money", given the scale of projects involving expeditions in the Arctic and ice breakers...

I understand this is a tough question, but any way you could always have a buffer of these in stock, and you would produce some more before you get completely empty? ...

jerabaul29, 2021-05-10:
PS: wondering if we should discuss this kind of questions by private email some day.

jerabaul29, 2021-06-08:
Hehe, I was too slow, the whole series of 10 you produced 2 weeks ago was sold out before I had the time to order some 😅 I do not have a large project now to buy many at once (hope this happens soon though), but trying to get to buy a couple when they are back in stock for some prototyping...

nseidle, 2021-06-09:
Sorry! I'll get more AGTs onto the schedule. A problem, but a great problem to have! Please place your orders so we can plan accordingly. Or email Customer Service for larger qty/lead times.

jerabaul29, 2021-06-25:
I am still waiting for confirmation if we get funding for a larger order or not, so I cannot really place a big order now, but I definitely want to buy a couple of units when they come back into stock. I know of at least a couple more colleagues who want to do the same. Looks like the passive antenna are out of stock too though... Would be amazing with fresh AGTs on the store and passive antenna to come with :) .

A challenge we have is that 1) it is difficult to know in advance when we get a "small" funding (and "small" can quite easily be as large as 10 AGTs and the equipment around them), 2) usually we have very little time between knowing that we have got some funding, and the deadline to ship the instruments to the people who deploy it. It means that we are quite dependent on finding the components buffered on your website to manage to have successful projects built on AGTs.

jerabaul29, 2021-09-16:
You are victim of your success, sold out again on your website ^^ :) . FYI: we (both me and some foreign colleagues) plan on buying a small round of 5 to 8 of these soon, and we hope to get the funding to buy another round of 10 to 20 in the course of the fall - if they are available again :) .

PaulZC, 2021-09-16:
Hi JR,
To quote Black Adder, we have a cunning plan... Watch this space!
Best wishes,
Paul

jerabaul29, 2021-09-21:
@PaulZC any information you can give us? If an AGT v2 would come out in a couple of months or so that would be very useful for us to know so that we do not "overfit" to an AGT v1 for example, but wait a bit for the new board :) . Or if there is no plan for AGT v2 in the foreseeable future, would be useful to know too, so that we can overfit agressively our design to it :) .

jerabaul29, 2021-09-21:
Sounds very good :) .

Just a small question or two: is it possible to get an idea of what you intend to tweak? May we expect more broken pins and / or an integrated step up step down, or other things? I am considering designing a "master PCB" on which to put my AGT and additional components, would be helpful to know if I should wait for working on it :) .

PaulZC, 2021-09-21:
The board dimensions and pin positions will stay the same. There will not be any additional pins this time around.
We are making small changes to the power circuit but the three power options will stay the same: USB, LiPo and Solar/AA/AAA.
Like I said, the main intention is to ensure the AGT remains in stock for customers like yourself.
BW, Paul

jerabaul29, 2021-09-21:
Ok, thanks :) .

jerabaul29, 2021-10-15:
(any update on this and the updated trackers? We have 2 orders for a total value of close to 20kUSD we would like to place, would be great to have some ideas about timeline :) ).

lewispg228, 2021-10-15:
Hey @jerabaul29 , 
I'd like to put you in touch with a dedicated account rep here at SparkFun. They would be happy to help you directly with these larger orders. 

Could you please email me directly at my email and I'll put you in touch? Mine is firstname.lastname@SparkFun.com Hopefully not too many spambots will catch that :) Looking forward to hearing from you. Thanks! -Pete Lewis

jerabaul29, 2021-11-24:
I have sent an email to SF sales putting you in copy @lewispg228 , if you have not received it let me know :) .

We have money for building around 20 AGT-based instruments :) . We hope to get some more money to build at least 20 more by mid 2022 (always unsure though if / when we get the money). At least 3 more groups we collaborate with are getting similar fundings and have plans to build some small series too. We also have some hope that one of our partners gets a larger funding and can build around 50 AGT-based systems. So if we and colleagues are a bit lucky with funding, we may need quite many boards withing 6 to 12 months. In worst case, if we are unlucky with our fundings, we will still need this typical number of boards anyways, but on the longer term (12 to 24 months instead +-).

We were discussing with some additional partners and trying to convince them to use AGTs. A challenge we have is to convince them that it is worth investing into R&D and development centered around the AGT. The worry we hear all the time is "the stocks are empty, and we cannot take the risk to not be able to perform a multi-million USD fieldwork because some components are  not available". So I think availability / making sure to always have a few tens of these in stock at all time would mean a lot for how many / which actors would take in use the AGT.

brhoff720, 2021-11-27:
Hi @jerabaul29,

While we are still subject to the ongoing supply chain challenges, the move of this product out of SparkX and becoming a SparkFun product, is directly related to making inventory availability more consistent. We are really excited about the product and the uses you and others are applying to.