sparkfunX/Artemis_Global_Tracker #28: RAM power down: what happens then?

jerabaul29, 2021-05-14:
I saw that there is some 'RAM power down' happening for low power sleep:

https://github.com/sparkfunX/Artemis_Global_Tracker/blob/ee5a23083b2e0c621d061a64d372bf5ff3b42a3b/Software/examples/Example15_BetterTracker/Example15_BetterTracker.ino#L1004

Does this mean that all the memory content outside of the first 512kb will be lost? (Assume it is b and not B as there are only 384kB of RAM?). Do you know how much power that saves?Asking as it sounds a bit scary; do we have any guarantees about how the RAM memory layout looks like? Any guarantees that the compiler does not use some RAM at the end and the start of the memory? What with dynamically allocated stuff? ...

If this is the wrong place for asking and I should ask on the main repo for Artemis just let me know :) .

jerabaul29, 2021-05-14:
Ahh no sorry, this is only flash as you point out, and you do not turn out RAM. Sorry for the noise, closing.

PaulZC, 2021-05-14:
Hi JR,
You have a valid point though. The AGT deep sleep code does not quite match the latest OpenLog Artemis code:

https://github.com/sparkfun/OpenLog_Artemis/blob/69682123ea2ae3ceb3983a7d8260eaef6259a0e2/Firmware/OpenLog_Artemis/lowerPower.ino#L142

I don’t think you will notice any difference, but I will update the AGT code to match.

All the best,
Paul

PaulZC, 2021-08-09:
Corrected in v2 - please see the revised examples for more details.
Best wishes,
Paul

closed 2021-08-09 by PaulZC