sparkfunX/Artemis_Global_Tracker #29: tracker suddenly failing; failing component? Failing solder? Other possible cause?

jerabaul29, 2021-05-15:
I have some AGTs doing some tests outside. One of them suddenly stopped to work after a couple of days and to transmit iridium messages.

After some debugging, I have reached the following conclusion:

- The code itself seems to work well; I uploaded the code on another AGT and it works fine
- To be on the safe side, I re-uploaded the code on the failing AGT, in the exact same configuration as the AGT that was working well above; still no success

I investigated in a bit more details the failure then:

- The failure is due to incapability to switch on the Iridium modem; I can actually see it, the 9603 power LED does not turn on
- I did a bit of serial debug analysis, and this got confirmed; the MCU does not manage to talk to the 9603 modem even after the supercap is charged and the driving pin is turned HIGH and everything (and I know this should work, as it works on the other AGT, while not on the failing one)
- I did one more test: shorting the 5V from the USB input into the power rail of the iridium modem (by feeding it on the left pin of the resistor upstream of the 9603N power LED); then the 9603 power LED fires up, and the AGT tracker works again fine...

So I think that there is some possibility that there is some form of hardware failure that prevents providing power to the iridium modem in my case. I see two main hypothesis:

- 1) the digital pin out of the Artemis driving the modem on is dead; I see no reason for this, the AGT was working and then suddenly stopped working, it was in a tight box outside, there was no possibility for shorting / power spike etc.

- 2) Somehow, a solder failed on the board. There are some temperature fluctuations outside of my home (maybe from +20 degrees C middle of day to a couple of - degrees at night these days). Is it possible that a solder somewhere on the PCB failed due to "temperature cycling", and that this broke some connection

Or any other idea what the cause can be?

- if you need to board to investigate I am willing to ship it to you
- we had a couple of inexplained failures during a deployment in the Arctic a few months ago, with AGTs suddenly stopping to work while testing on the deck of our boat; wondering if it could be some similar cause to this one?

Any more ideas / thoughts?

jerabaul29, 2021-05-15:
PS: the failure we had with at least one of the instruments on the boat was failure to talk to the GPS module; not sure what the other failures were.

PaulZC, 2021-05-16:
Hi JR,

If your AGT is inside an insulated box, I wonder if the temperature fluctuations are possibly larger than ~20 degrees? Repeated thermal cycling can certainly cause poor solder joints to weaken and fail.

Please check the Artemis digital pin 22 signal. If wonder if the joint under the Artemis has failed?

[image]

Unfortunately R14 is hidden by the modem. If you remove the modem, you will be able to probe the left end of R14, just to see if if goes high at the expected time. Or you could carefully scrape the solder resist off a small area of track on the rear of the board and probe there instead.

[image]

Let me know what you find.

Best wishes,
Paul

jerabaul29, 2021-08-02:
I was a bit busy but got the time to get back to this one today. Strangely, I tried the exact same thing, and it worked this time. Looks like there is some intermittent problem happening.

I think the most likely option is a cracked solder or something like this. I will not use this AGT for real world deployment, will just keep it home for testing, and report if I notice more weirdness.

We could either close or keep open this issue, as you like it best. Would be curious to have one issue of this kind open, so that if other users observer similar issues, they can report it.

PaulZC, 2021-08-02:
Hi JR,
Glad it is working again for you. It could be an intermittent solder joint.
Yes, please leave this issue open so we can see if anyone else has the same issue.
Best wishes,
Paul

jerabaul29, 2021-08-25:
A bit of updating here. I do have one AGT that is a really pain with intermittent issues, probably some bad soldering contacts, and one other that has had an issue now and then but mostly works fine, but I suspect the same. This is out of about 15 AGTs I have been spending time with. I am quite confident this is not a software issue, as some of the AGTs work like clocks.

I am a bit surprised by this relatively high "failure rate", I had not had such issues before. Of course I may just be unlucky, and / or other things (unlucky with a static electricity discharge or similar) may have happened. Still, a bit scaring when I want to use these in mission critical contexts, in an environment where there are quite large temperature fluctuations.

I was looking at the boards, thinking about their design, and wondering about the following. How "sound" / common / good or bad practice is it to have two large area PCBs with many soldering points between them being soldered on top of each other? This is more or less how the Artemis red "nail size" PCB is put on top of the main AGT PCB, right?

[image]

Just wondering, if this does is a potential point of weakness in the hardware robustness, any way future designs could see the Ambiq MCU and stuff around soldered directly on the main PCB, and / or some way during the manufacturing to make really sure that the nail size PCB gets really reliably soldered, without gas bubbles, weak solders that may crack with temperature fluctuations etc?

What is your opinion on this? I know quite little about hardware and industrial processes during manufacture, but curious to hear your thoughts :) .

tnn77, 2021-12-02:
Dear Sparkfun, 

We have purchased several AGTs in the last year or so to work with @jerabaul29 's design. 
We have had a couple of faulty QWIIC connectors and wish to report them. 

**S/N 203756:**
the QWIIC connector simple came off the board as you can see in the picture. I actually don't know when this happened, but I don't recall treating the AGT in a rough manner. Nevertheless, we wish to use AGTs in Jean's ocean measuring device, usually deployed in harsh conditions, so the solder quality is a bit of a concern. 

Note this was purchased a while ago, but did not come in use for months.

[image]

S/N 203767
Regarding this AGT, we used this straight away and noticed that the QWIIC port did not work. When we used the SDA and SCL pins, it worked. Visually, I cannot see anything wrong. 

We'd be grateful if you could advise us to diagnose the problem.

[image]

Thank you. 
Tak