sparkfunX/Artemis_Global_Tracker #31: getting the DS18B20 to work with the AGT

jerabaul29, 2021-07-02:
Today I am struggling to get the DS18B20 to work with the Artemis boards. I have managed to get it to work with the RedBoard, and it seems that getting it to work is quite sensitive on the value of the pullup resistor, and that the value to pick depends on the pins. See:

https://github.com/sparkfun/Arduino_Apollo3/issues/409

I am trying to get it to work on the Artemis Global Tracker now. I am not having success so far.

What I am using:

- Artemis Global Tracker, being programmed as an "Artemis module" as usual for this board
- AGT GND to DS18B20 black
- AGT 3.3V to DS18B20 red
- AGT D4 to DS18B20 signal (also tried AGT D35)
- 4.7kOhm pullup between the DS18B20 red and signal

I am using the same kind of sketch I was successful with on the RedBoard (see under). I am not able to detect the DS18B20 (only getting some "No more addresses" prints, i.e. cannot find the thermometer).

Any idea what may cause that?

```
#include <OneWire.h>

// OneWire DS18S20, DS18B20, DS1822 Temperature Example
//
// http://www.pjrc.com/teensy/td_libs_OneWire.html
//
// The DallasTemperature library can do all this work for you!
// https://github.com/milesburton/Arduino-Temperature-Control-Library

OneWire  ds(4);  // on pin 10 (a 4.7K resistor is necessary)

void setup(void) {
  Serial.begin(9600);
}

void loop(void) {
  byte i;
  byte present = 0;
  byte type_s;
  byte data[12];
  byte addr[8];
  float celsius, fahrenheit;
  
  if ( !ds.search(addr)) {
    Serial.println("No more addresses.");
    Serial.println();
    ds.reset_search();
    delay(250);
    return;
  }
  
  Serial.print("ROM =");
  for( i = 0; i < 8; i++) {
    Serial.write(' ');
    Serial.print(addr[i], HEX);
  }

  if (OneWire::crc8(addr, 7) != addr[7]) {
      Serial.println("CRC is not valid!");
      return;
  }
  Serial.println();
 
  // the first ROM byte indicates which chip
  switch (addr[0]) {
    case 0x10:
      Serial.println("  Chip = DS18S20");  // or old DS1820
      type_s = 1;
      break;
    case 0x28:
      Serial.println("  Chip = DS18B20");
      type_s = 0;
      break;
    case 0x22:
      Serial.println("  Chip = DS1822");
      type_s = 0;
      break;
    default:
      Serial.println("Device is not a DS18x20 family device.");
      return;
  } 

  ds.reset();
  ds.select(addr);
  ds.write(0x44, 1);        // start conversion, with parasite power on at the end
  
  delay(1000);     // maybe 750ms is enough, maybe not
  // we might do a ds.depower() here, but the reset will take care of it.
  
  present = ds.reset();
  ds.select(addr);    
  ds.write(0xBE);         // Read Scratchpad

  Serial.print("  Data = ");
  Serial.print(present, HEX);
  Serial.print(" ");
  for ( i = 0; i < 9; i++) {           // we need 9 bytes
    data[i] = ds.read();
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  Serial.print(" CRC=");
  Serial.print(OneWire::crc8(data, 8), HEX);
  Serial.println();

  // Convert the data to actual temperature
  // because the result is a 16 bit signed integer, it should
  // be stored to an "int16_t" type, which is always 16 bits
  // even when compiled on a 32 bit processor.
  int16_t raw = (data[1] << 8) | data[0];
  if (type_s) {
    raw = raw << 3; // 9 bit resolution default
    if (data[7] == 0x10) {
      // "count remain" gives full 12 bit resolution
      raw = (raw & 0xFFF0) + 12 - data[6];
    }
  } else {
    byte cfg = (data[4] & 0x60);
    // at lower res, the low bits are undefined, so let's zero them
    if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
    else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
    else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
    //// default is 12 bit resolution, 750 ms conversion time
  }
  celsius = (float)raw / 16.0;
  fahrenheit = celsius * 1.8 + 32.0;
  Serial.print("  Temperature = ");
  Serial.print(celsius);
  Serial.print(" Celsius, ");
  Serial.print(fahrenheit);
  Serial.println(" Fahrenheit");
}
```

jerabaul29, 2021-07-02:
I tried for several hours on this, and in the end I gave up and grabbed a DS18B20 from another manufacturer - and that worked out of the box.

So it seems that some manufacturers sell low quality DS18B20. Weird, I bought the ones that work badly from Sparkfun. The ones from Adafruit seem to work nicely. Closing.

jerabaul29, 2021-07-02:
Actually, some more strange stuff:

- I have 1 DS18B20 sensor that works fine with the Artemis
- I have another DS18B20 that works fine with the Redboard, but no ways to get it to work with the Artems

Are there some pullups / pulldowns of some kind that I may not be aware of? I looked at the schematics but could not find any.

PaulZC, 2021-07-02:
Hi JR,
Please don't raise the same question in two places.
Closing in favour of [this issue](https://github.com/sparkfun/Arduino_Apollo3/issues/409).
Best wishes,
Paul

closed 2021-07-02 by PaulZC

jerabaul29, 2021-07-02:
Well, actually I got it to work on the Artemis Reboard, but I do not manage to get the exact same sensor to work with the AGT. This is why I opened a new issue here: I wonder if this is AGT specific.